ArduCAM/Arducam-Pivariety-V4L2-Driver #1: UC 572

neil-benn, 2021-05-26:
Hello,

  I've just installed this driver and plugged a UC-572 into a raspberry pi4.  However I cannot see any activity with dmesg to see if anything is installed, no new video devices appear and nothing happens with cheese or vlc.  I simply went to the release folder and ran sudo ./install.sh and I was asked to reboot. 

  Can you advise me as to what I'm doing wring please?

Regards,

NEil

AllaboutJay, 2021-05-27:
Hello, please follow our documentation to install the driver, please follow our documentation to access the camera

AllaboutJay, 2021-05-27:
Hello，

Please follow the documentation to install the driver https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-install-kernel-driver-for-pivariety-camera/,
and follow the documentation to access the camera https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-access-and-program-the-pivariety-camera-using-command-line-python-opencv-and-gstreamer/

Thank you！

Regards,
Xin

AllaboutJay, 2021-05-27:
> Hello,
>
> I've just installed this driver and plugged a UC-572 into a raspberry pi4. However I cannot see any activity with dmesg to see if anything is installed, no new video devices appear and [...]

Hello，

Please follow the documentation to install the driver https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-install-kernel-driver-for-pivariety-camera/,
and follow the documentation to access the camera https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-access-and-program-the-pivariety-camera-using-command-line-python-opencv-and-gstreamer/

Thank you！

Regards,
Xin