ArduCAM/Arducam-Pivariety-V4L2-Driver #2: No driver detection

kamilexg, 2021-05-28:
Hello,
I have Raspberry Pi 3b+. I have followed this guide: https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-install-kernel-driver-for-pivariety-camera/
After installation I tried to detect the driver. Here is the result:
[image]
](url)
After a while I got errors:
[image]
[image]
[image]
[image]
[image]
[image]
After these errors, I checked the drivers again and got these results:
[image]
Please help explain what I am doing wrong.

Regards,

kamilexg

kamilexg, 2021-06-02:
Now i got this if retry installation driver.
[image]
I have a damaged camera?

mwkldeveloper, 2021-06-03:
I have the same problem, the camera never be detected.

kamilexg, 2021-06-03:
I have a solution. Most likely I didn't have the camera enabled in raspi config. For this I could not install the drivers correctly from this website:
https://www.arducam.com/docs/arducam-obisp-mipi-camera-module/3-use-on-raspberry-pi/3-0-emergent-temporary -driver-for-the-lastest-rpi-2020-12-02-update/

closed 2021-06-03 by kamilexg

AllaboutJay, 2021-06-07:
> Now i got this if retry installation driver.
> [image]
> I have a damaged camera?

Hello, 

We recommend you follow the DOC:
https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/unboxing-video-pivariety-camera/
Note: Please be sure to download the mirror using the link we provided.

and if you still report the error, please send me.

Thanks!