ArduCAM/Arducam-Pivariety-V4L2-Driver #4: please compile for the new version of raspberry pi OS kernel 5.10.52 !

MarcBresson, 2021-08-09:


glddiv, 2021-08-10:
Hi @MarcBresson 
Currently we only provide the pre-compiled kernel driver of the official SD Card image.
Because it is relatively fixed (it may be updated every few months), but you have too many kernel versions to upgrade using rpi-update, we cannot provide pre-compiled drivers for each version.

Since we provide the source code of the kernel driver, I suggest you compile the driver of the kernel version you need with our documentation.
https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-install-kernel-driver-for-pivariety-camera/#ftoc-heading-10

MarcBresson, 2021-08-10:
The kernel 5.10.52 is actually the new default raspberry pi version. It is no longer 5.10.17

MarcBresson, 2021-08-10:
furthermore, i do have an issue with the driver compilation. libcamera-vid is working fine, but libcamera-still gives me an error (full thread here https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=317401)

glddiv, 2021-08-10:
Hi @MarcBresson 

Thank you for the information,
Regarding libcamera-still, can you try to run libcamera-still locally instead of using ssh?
Or execute the following command:
`libcamera-still -t 0 --viewfinder-width 1920 --viewfinder-height 1080`

MarcBresson, 2021-08-10:
Executing the command locally still gives me the same error.
Furthermore i'm running a headless raspberry pi, so i guess viewfinder isn t doing anything (yet, this command did not generat any rror, but was just doing nothing)

glddiv, 2021-08-12:
Hi @MarcBresson

This seems to be normal behavior, and the function of this command is to preview the image indefinitely...it doesn't do anything else.
If you want to take a photo, you can execute:
libcamera-still -t 5000 -o test.jpg --viewfinder-width 1920 --viewfinder-height 1080

-t 5000 means wait for 5s (for preview and ISP adjustment parameters) you can reduce it.
-o test.jpg means to output an image named test.jpg.