ArduCAM/Arducam-Pivariety-V4L2-Driver #5: Document which libcamera commit ID corresponds to the binary packages

pinchartl, 2021-08-19:
The Pivariety documentation links to pre-built libcamera binaries, and also mentions that libcamera can alternatively be built from sources (https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-access-and-program-the-pivariety-camera-using-command-line-python-opencv-and-gstreamer/#12-build-from-the-source-code). As libcamera evolves quite quickly and doesn't have a stable ABI yet, building from sources needs to match the sources against which ipa_rpi.so was built. Could you please indicate which commit ID in the upstream libcamera git tree corresponds to the binary packages ? It would also be nice to include that information directly into new releases, possibly in the Debian package itself.

pinchartl, 2021-09-21:
Ping. Providing the commit ID for the binary packaged would be a simple way to comply with the LGPL license.

jonjonarnearne, 2022-02-10:
It looks like they did it for for v0.0.3:
https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/tag/ipa-v0.0.3