ArduCAM/Arducam-Pivariety-V4L2-Driver #7: IMX230 green center

Schaggo, 2021-09-20:
I adapted an MFT lens to the sensor - which produces incredibly sharp images but features a green smudge in the center of the image. Does the driver/stack add some color correction for the original lens?

ArduCAM, 2021-09-21:
It is not good to add a custom lens on the IMX230 camera, the default camera driver is calibrated with the stock lens.

Schaggo, 2021-09-21:
It is clearly possible to change the lens shading table. Your own webpage features a how-to for a different sensor in combination with raspistill.

Schaggo, 2021-09-21:
Sorry, I just realized you use your own IPA.

ArduCAM, 2021-09-21:
Yes, it is a totally different architecture compared to raspistill command, the webpage is only valid for raspistill with native RPI cameras.

closed 2023-03-14 by Schaggo