ArduCAM/Arducam-Pivariety-V4L2-Driver #8: kernel 5.10.63

tityann, 2021-09-23:
hello
could you please add support for kernel 5.10.63
thanks

UnexpectedMaker, 2021-12-20:
Bumping this please.... just got my B0324 and it's not compatible with bullseye :(
My kernel version is...
`Linux aoi2 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux`

Alternately, please provide steps to compile from source. Thanks!

UnexpectedMaker, 2021-12-20:
I've now tried adding support myself using both of your suggested ways... patching and building source from headers, and after both attempts, I still get the same issue...

```pi@aoi2:~/arducam/Arducam-Pivariety-V4L2-Driver/src $ dmesg | grep arducam
[    8.206317] arducam: loading out-of-tree module taints kernel.
[    8.216337]  arducam-pivariety: arducam_read: Reading register 0x103 failed
[    8.216349] arducam-pivariety 10-000c: probe failed
[   32.476542] arducam_vddl: disabling
[   32.476565] arducam_vdig: disabling
[   32.476593] arducam_vana: disabling```

Maybe your instructions need to change now for bullseye? maybe something is different now?