ArduCAM/Arducam-Pivariety-V4L2-Driver #9: 64 bit kernel support

rosterloh, 2021-11-08:
Are you planning to support the 64 bit kernel (5.10.63-v8+) in your releases at any point?

rebeccaRossRobotics, 2021-11-22:
I've just tried to build the driver based off your pull request as I'm using the latest 64bit image also. 
I'm following section 2.2 on [this](https://www.arducam.com/docs/cameras-for-raspberry-pi/pivariety/how-to-install-kernel-driver-for-pivariety-camera/#ftoc-heading-15) webpage but having the git pointing at your branch and am getting the following error after running `make && sudo make install`.

```
make -C /lib/modules/'uname -r'/build M=$PWD
make[1]: Entering directory '/home/pi/5.10.63-v8+/linux-abff5012ab7ccbe1227b9d0af69697cc592df44c'
  CC [M]  /home/pi/Arducam-Pivariety-V4L2-Driver/src/arducam.o
/home/pi/Arducam-Pivariety-V4L2-Driver/src/arducam.c:1196:12: warning: ‘arducam_add_extension_pixformat’ defined but not used [-Wunused-function]
 1196 | static int arducam_add_extension_pixformat(struct arducam *priv)
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  MODPOST /home/pi/Arducam-Pivariety-V4L2-Driver/src/Module.symvers
ERROR: modpost: "cpu_hwcap_keys" [/home/pi/Arducam-Pivariety-V4L2-Driver/src/arducam.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/home/pi/Arducam-Pivariety-V4L2-Driver/src/arducam.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:124: /home/pi/Arducam-Pivariety-V4L2-Driver/src/Module.symvers] Error 1
make[2]: *** Deleting file '/home/pi/Arducam-Pivariety-V4L2-Driver/src/Module.symvers'
make[1]: *** [Makefile:1729: modules] Error 2
make[1]: Leaving directory '/home/pi/5.10.63-v8+/linux-abff5012ab7ccbe1227b9d0af69697cc592df44c'
make: *** [Makefile:10: default] Error 2
```

Any ideas on how to fix it? Or can you tell me how to build the driver successfully on the latest 64bit kernel

rosterloh, 2021-11-23:
I built the kernel from these
<https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building-the-kernel-locally>
docs. After cloning I checked out the 1.20211029 as `sudo dpkg -l | grep
raspberrypi-kernel` shows the tag of the running kernel. I then applied the
patches to this. Hope this helps.

mgvs, 2022-02-11:
Dears any updates with 64-bit (bullseye) support?

rosterloh, 2022-02-11:
@mgvs they didn't merge this version but there is support for the current kernel version (5.10.92-v8+ from tag 1.20220120-1) in their kernel_driver_links.txt. So you can install using the [install_pivariety_pkgs.sh](https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/tag/install_script) script

PR #10 (2022-07-05): Add current 64 bit kernel module support
Fixes #9. Tested and working on Pi 4 64 bit Bullseye
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
```
```
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
```
pi@raspberrypi:~ $ dmesg | grep arducam
[    7.334977] ar [... cut]
closed 2022-07-05 by ArduCAM