ArduCAM/Arducam-Pivariety-V4L2-Driver #20: focuser.py : OSError: [Errno 25] Inappropriate ioctl for device

deepcoder, 2022-03-16:
with:
Arducam 16MP Autofocus Raspberry Pi Camera Module

I am getting the following error when trying to run your autofocus example.
```
Traceback (most recent call last):
  File "/home/pi/v4l2_utils.py", line 64, in get_device_controls_by_class
    fcntl.ioctl(fd, v4l2.VIDIOC_QUERYCTRL, queryctrl)
OSError: [Errno 25] Inappropriate ioctl for device

During handling of the above exception, another exception occurred:
```

kbingham, 2022-04-01:
That means the RPi kernel is not using a patched driver for the VCM. I have this working locally, and Arducam also provide the fix (I think in this repository). I'll be working towards getting this integrated in the RPi kernel and upstream too - but I don't know when I'll complete that.

kbingham, 2022-04-01:
In fact, I fear 'Inappropriate ioctl' might also be that you are trying to operate the focuser on the wrong device... Have  you patched your device tree and kernel sources for the RPi?

arogozhnikov, 2022-08-05:
I have same problem with 64mp autofocus camera. 
I've followed all instructions to install patch to kernel and dt overlay. Autofocusing works on camera, but the code provided in the repo does not

arogozhnikov, 2022-08-10:
actually @kbingham above is correct: I used the wrong device, and the correct one was /dev/v4l-subdev1 for arducam64mp

kbingham, 2022-08-10:
Glad you resolved it. I hope we'll get the focus support for these devices merged in libcamera which will make sure it gets the right device to prevent this confusion in the future.