ArduCAM/Arducam-Pivariety-V4L2-Driver #27: Repository is broken: libcamera-apps:arm64 (= 0.0.8) has no Size information

wcbonner, 2022-08-17:
Attempting to update the drivers for my 64MP camera module gets me an error and I'm not sure I'm on the latest versions. 

```
wim@WimPi4-Dev:~ $ sudo ./install_pivariety_pkgs.sh -p libcamera_apps
=================================================
Hardware Revision: d03114
Kernel Version: 5.15.56-v8+
OS Codename: bullseye
ARCH: aarch64
=================================================

kernel:5.15.56-v8+
--2022-08-17 09:02:15--  https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/libcamera_apps_links.txt
Resolving github.com (github.com)... 192.30.255.113
Connecting to github.com (github.com)|192.30.255.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/051154c5-72db-4912-a8d1-1b44f132c828?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220817%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220817T160216Z&X-Amz-Expires=300&X-Amz-Signature=ae8b65fe3db39ad4c04151426b6722072dfba2479d92050fc8d73b23891d78dd&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera_apps_links.txt&response-content-type=application%2Foctet-stream [following]
--2022-08-17 09:02:16--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/051154c5-72db-4912-a8d1-1b44f132c828?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220817%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220817T160216Z&X-Amz-Expires=300&X-Amz-Signature=ae8b65fe3db39ad4c04151426b6722072dfba2479d92050fc8d73b23891d78dd&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera_apps_links.txt&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.110.133, 185.199.109.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2231 (2.2K) [application/octet-stream]
Saving to: ‘libcamera_apps_links.txt’

libcamera_apps_links.txt                              100%[=========================================================================================================================>]   2.18K  --.-KB/s    in 0s      

2022-08-17 09:02:16 (11.1 MB/s) - ‘libcamera_apps_links.txt’ saved [2231/2231]

--2022-08-17 09:02:16--  https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/libcamera-apps-v0.0.5/libcamera-apps-0.0.8-bullseye-arm64.deb
Resolving github.com (github.com)... 192.30.255.113
Connecting to github.com (github.com)|192.30.255.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/2ab9419a-7abf-482a-9baa-fdb53f85fe6f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220817%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220817T160216Z&X-Amz-Expires=300&X-Amz-Signature=56e11d45efc88449000ef03124b6d27a00037be2c8f271659724a73f95ed0e9a&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera-apps-0.0.8-bullseye-arm64.deb&response-content-type=application%2Foctet-stream [following]
--2022-08-17 09:02:16--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/2ab9419a-7abf-482a-9baa-fdb53f85fe6f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220817%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220817T160216Z&X-Amz-Expires=300&X-Amz-Signature=56e11d45efc88449000ef03124b6d27a00037be2c8f271659724a73f95ed0e9a&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera-apps-0.0.8-bullseye-arm64.deb&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 283496 (277K) [application/octet-stream]
Saving to: ‘libcamera-apps-0.0.8-bullseye-arm64.deb’

libcamera-apps-0.0.8-bullseye-arm64.deb               100%[=========================================================================================================================>] 276.85K  --.-KB/s    in 0.02s   

2022-08-17 09:02:16 (11.5 MB/s) - ‘libcamera-apps-0.0.8-bullseye-arm64.deb’ saved [283496/283496]

Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]            
Get:3 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:4 http://archive.raspberrypi.org/debian bullseye InRelease [23.7 kB]           
Get:5 http://security.debian.org/debian-security bullseye-security/main armhf Packages [173 kB]
Get:6 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [176 kB]
Get:7 http://security.debian.org/debian-security bullseye-security/main Translation-en [111 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [295 kB]
Get:9 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [287 kB]
Get:10 http://archive.raspberrypi.org/debian bullseye/main arm64 Contents (deb) [629 kB]
Get:11 http://archive.raspberrypi.org/debian bullseye/main armhf Contents (deb) [1,089 kB]
Fetched 2,874 kB in 3s (966 kB/s)                                        
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
6 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libcamera-apps' instead of './libcamera-apps-0.0.8-bullseye-arm64.deb'
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 6 not upgraded.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/wim/libcamera-apps-0.0.8-bullseye-arm64.deb libcamera-apps arm64 0.0.8 [283 kB]
(Reading database ... 102126 files and directories currently installed.)
Preparing to unpack .../libcamera-apps-0.0.8-bullseye-arm64.deb ...
Unpacking libcamera-apps (0.0.8) over (0.0.8) ...
Setting up libcamera-apps (0.0.8) ...
W: Repository is broken: libcamera-apps:arm64 (= 0.0.8) has no Size information
```

mikeprevette, 2022-09-05:
Same issue.

evanhelene, 2022-09-10:
same for armhf

PatrickFab21, 2023-06-04:
Same issue.