ArduCAM/Arducam-Pivariety-V4L2-Driver #29: Arducam 64Mp Hawkeye never worked for release

FPSychotic, 2022-09-18:
From my camera module arrived never worked
I followed all the mixed and redundant tutorials you have, with the rpi os .img  from the links you given , both and this still doesnt work.
Can you put here a link to the the exact tutorial and exact OS version that I need to follow?

What about ubuntu, has it support , if not will have it?

FPSychotic, 2022-09-19:
from a installation clean, and a guide updated one week ago. I have a Rpi0 2 W and followed that section , but every other, and every tutorial, and every OS.
ANd you didnt answer here any question from users.

```
pi@raspberrypi:~ $ wget -O install_pivariety_pkgs.sh https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/install_pivariety_pkgs.sh
--2022-09-19 02:50:00--  https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/install_pivariety_pkgs.sh
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/4eff0def-1082-43f8-848d-442848383038?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220919%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220919T015000Z&X-Amz-Expires=300&X-Amz-Signature=85d4cf15411ecb4d7f9907c76e314f6d7a55446c7b2ff49a0956f8ce7500a269&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dinstall_pivariety_pkgs.sh&response-content-type=application%2Foctet-stream [following]
--2022-09-19 02:50:01--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/4eff0def-1082-43f8-848d-442848383038?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220919%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220919T015000Z&X-Amz-Expires=300&X-Amz-Signature=85d4cf15411ecb4d7f9907c76e314f6d7a55446c7b2ff49a0956f8ce7500a269&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dinstall_pivariety_pkgs.sh&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.110.133, 185.199.108.133, 185.199.111.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10244 (10K) [application/octet-stream]
Saving to: ‘install_pivariety_pkgs.sh’

install_pivariety_p 100%[===================>]  10.00K  --.-KB/s    in 0.002s  

2022-09-19 02:50:01 (5.33 MB/s) - ‘install_pivariety_pkgs.sh’ saved [10244/10244]

pi@raspberrypi:~ $ chmod +x install_pivariety_pkgs.sh
pi@raspberrypi:~ $ ./install_pivariety_pkgs.sh -p 64mp_pi_hawk_eye_kernel_driver 
=================================================
Hardware Revision: 902120
Kernel Version: 5.15.61-v7+
OS Codename: bullseye
ARCH: armv7l
=================================================

--2022-09-19 02:50:46--  https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/packages.txt
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/ba2ae6e6-a95e-4313-aa07-1cd7d1f7e687?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220919%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220919T015047Z&X-Amz-Expires=300&X-Amz-Signature=b76d102ccce3c55f8f3805a677d274a4170fb6a02f9975e9401e2905c809ab00&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dpackages.txt&response-content-type=application%2Foctet-stream [following]
--2022-09-19 02:50:47--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/ba2ae6e6-a95e-4313-aa07-1cd7d1f7e687?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220919%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220919T015047Z&X-Amz-Expires=300&X-Amz-Signature=b76d102ccce3c55f8f3805a677d274a4170fb6a02f9975e9401e2905c809ab00&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dpackages.txt&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.109.133, 185.199.111.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1360 (1.3K) [application/octet-stream]
Saving to: ‘packages.txt’

packages.txt        100%[===================>]   1.33K  --.-KB/s    in 0s      

2022-09-19 02:50:47 (5.33 MB/s) - ‘packages.txt’ saved [1360/1360]

kernel:5.15.61-v7+
--2022-09-19 02:50:48--  https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/64mp_pi_hawk_eye_kernel_driver_links.txt
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/420f2754-cd4f-4eb7-a309-bdfac7d3e214?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220919%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220919T015048Z&X-Amz-Expires=300&X-Amz-Signature=0206dd9062d9ceb9eba26395b88d754a6a72c3c6e711dbaed7ca5ee089b4cb65&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20 [... cut ...]
Release/bin/5.15.61-v8+/arducam_64mp.ko
Release/bin/5.15.61-v8+/dw9807-vcm.ko
Release/install_driver.sh
Release/uninstall_driver.sh
Installing DW9807 and ARDUCAM_64MP_EEPROM and ...
--------------------------------------
Your camera is ,and the relevant drivers have been installed.
pi@raspberrypi:~ $ dmesg | grep arducam
pi@raspberrypi:~ $ dmesg | grep arducam
pi@raspberrypi:~ $ dmesg | grep arducam
pi@raspberrypi:~ $ ls /dev/video0
ls: cannot access '/dev/video0': No such file or directory
pi@raspberrypi:~ $ 

```

stuartcol, 2022-11-07:
> from a installation clean, and a guide updated one week ago. I have a Rpi0 2 W and followed that section , but every other, and every tutorial, and every OS.
>
> ANd you didnt answer here any question [...]

My experience is that if you make a sad card with the raspberry pi 64bit image and then follow the Arducam instructions the camera works fine.

BUT .... as soon as you do a upgrade of the raspberry pi image it breaks the Arducam installation giving you the. "ls: cannot access '/dev/video0': No such file or directory". Error

stuartcol, 2022-11-08:
Apart from these issues, it is a nice camera

FPSychotic, 2022-11-08:
No, is too dark and have not exposure control. It has something called
gain, but won't change nothing except if you rise it quite a lot but it
will generate a lot of noise. I don't think the videos are some with it, or
they were heavily modified,  as it is so dark that cannot make such
colours. No support here.
I made a microscope with it, was the most reasonable use for it.