ArduCAM/Arducam-Pivariety-V4L2-Driver #30: v4l2-ctl missing controls (16MP Autofocus Camera)

giano574, 2022-09-19:
When executing `v4l2-ctl --list-ctrls -d /dev/v4l-subdev1` only `focus_absolute` is listed as a control. I need to be able to adjust shutter speed and gain on the fly. Using `libcamera-vid --shutter x --gain x` works, so why can't I use v4l2-ctl to adjust it?

kbingham, 2022-09-19:
When libcamera is in control of the camera, it is responsible for handling the gain and exposure.
What is your use case?

giano574, 2022-09-20:
I'm creating a video stream using [rtsp-simple-server](https://github.com/aler9/rtsp-simple-server), which uses libcamera (not apps) directly. I need to be able to adjust shutter speed and gain on the fly, and since `v4l2-ctl -d /dev/v4l-subdev1 -c focus_absolute=850` works, I thought I would be able to also adjust the shutter speed and gain in the same way.
Apparently `/dev/v4l-subdev0` is the device that exposes everything other than `focus_absolute` but changing `exposure` and `analogue_gain` doesn't work, I assume because - as you said - libcamera is in control.

kbingham, 2022-09-20:
I would say this should be an issue on the rtsp-simple-server then. Can you reference it here when you create it so I can follow it please?

giano574, 2022-09-20:
https://github.com/aler9/rtsp-simple-server/issues/1159
Is it documented somewhere in libcamera docs what one would have to do to change camera controls in the least "intrusive" way? From what I could gather from reading the doxygen docs, it seems like one would at least have to call `stop()` on the `libcamera::Camera` and then call `start` again with an updated `libcamera::ControlList`. Is that the case?

kbingham, 2022-09-20:
It doesn't have to stop and restart no - the request which are queued to get the frames can supply a control list as well. The desired controls should be part of that list which is submitted. That request is submitted by the rtsp-simple-server in the use case you've described, so rtsp-simple-server needs to add the controls.