ArduCAM/Arducam-Pivariety-V4L2-Driver #38: libcamera patch does not include arducam_64mp.json

gkrasin, 2022-12-20:
I am currently trying to debug an issue with getting full-res RAW10 photos from B0402 https://www.arducam.com/product/64mp-quad-camera-kit/ kit. For that, I need to be able rebuilt libcamera and libcamera-apps with Arducam patches. Apparently, the libcamera patch published at https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases does not include arducam_64mp.json that can be found in the https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/libcamera-v0.0.5/libcamera-dev-0.0.10-bullseye-arm64.deb

Potentially, this is because the last libcamera patch was published in July 2022, but the latest libcamera-dev binary is from October 2022. It's also quite probable that some other changes are not included into the published patch as well. It would be nice to see the latest libcamera patch published. Hopefully, it will allow me to debug & fix the issue that I am currently facing, which would unblock more orders from my team and (ideally) the potential fix could benefit all future Arducam B0402 users.

Thank you in advance.

havoc-dlt, 2023-01-03:
I'd like a general patch with all the necessary data contained too, so that I could possibly update it to other libcamera versions myself. Even if that means that said patch contains binary blobs. Would not be nice, but way better than it is now!

yaazzz, 2023-02-19:
> I'd like a general patch with all the necessary data contained too, so that I could possibly update it to other libcamera versions myself. Even if that means that said patch contains binary blobs. Would not be nice, but way better than it is now!
> 
> Or am I simply missing the place where to find that?

Any update on that one as well? Would be interesting to get this moving forward.

AryanMehta-xd, 2024-10-01:
@yaazzz @havoc-dlt @gkrasin  is this error resolved in the latest update ?
cause i am facing the same issue