ArduCAM/Arducam-Pivariety-V4L2-Driver #41: Can't install pivariety on NVIDIA Jetson Nano B01.

jasonrichdarmawan, 2023-02-04:
My goal is to use `libcamera`. **The problem is that I can't install libcamera on NVIDIA Jetson Nano B01.**

**What I've did (without libcamera):**
1. Follow https://github.com/ArduCAM/Jetson_IMX519_Focus_Example and set the `gstreamer_pipeline(capture_width=2328, capture_height=1748)` in JetsonCamera.py and `Camera(width=1280, height=1280)` in main.py. **The problem is that the saved image have black screen on the bottom.**

[image]

The output
```
werate-01@werate-01:~/NVIDIA_Jetson_Nano_Arducam_IMX519$ python3 main.py -i 7
nvbuf_utils: Could not get EGL display connection
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 4656 x 3496 FR = 10.000000 fps Duration = 100000000 ; Analog Gain range min 1.000000, max 16.000000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3840 x 2160 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 16.000000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 16.000000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 16.000000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 1 
   Output Stream W = 3840 H = 2160 
   seconds to Run    = 0 
   Frame Rate = 21.000000 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
```

**What I've did (trying to install libcamera):**
1. Follow the **Operating The Camera** section. However, it does not want to be installed.

The output
```
werate-01@werate-01:~/libcamera$ wget -O install_pivariety_pkgs.sh https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/install_pivariety_pkgs.sh
--2023-02-04 13:06:29--  https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/install_pivariety_pkgs.sh
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/143f4425-f176-48d6-b4b6-34407e6d0727?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230204%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230204T060629Z&X-Amz-Expires=300&X-Amz-Signature=5e68880d9359b599b25ccc5ebb40fc508f204a3a7a8709c9aacd3b38e3ef94a8&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dinstall_pivariety_pkgs.sh&response-content-type=application%2Foctet-stream [following]
--2023-02-04 13:06:29--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/143f4425-f176-48d6-b4b6-34407e6d0727?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230204%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230204T060629Z&X-Amz-Expires=300&X-Amz-Signature=5e68880d9359b599b25ccc5ebb40fc508f204a3a7a8709c9aacd3b38e3ef94a8&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dinstall_pivariety_pkgs.sh&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10610 (10K) [application/octet-stream]
Saving to: ‘install_pivariety_pkgs.sh’

install_pivariety_p 100%[===================>]  10,36K  --.-KB/s    in 0,002s  

2023-02-04 13:06:30 (4,73 MB/s) - ‘install_pivariety_pkgs.sh’ saved [10610/10610]

werate-01@werate-01:~/libcamera$ chmod +x install_pivariety_pkgs.sh
werate-01@werate-01:~/libcamera$ sudo apt update
[sudo] password for werate-01: 
Get:1 file:/var/cuda-repo-l4t-10-2-local  InRelease
Ign:1 file:/var/cuda-repo-l4t-10-2-local  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease
Ign:3 file:/var/visionworks-sfm-repo  InRelease
Get:4 file:/var/visionworks-tracking-repo  InRelease
Ign:4 file:/var/visionworks-tracking-repo  InRelease
Get:5 file:/var/cuda-repo-l4t-10-2-local  Release [564 B]
Get:6 file:/var/visionworks-repo  Release [2.001 B]
Get:5 file:/var/cuda-repo-l4t-10-2-local  Release [564 B]
[... 55 lines cut ...]

2023-02-04 13:07:57 (5,32 MB/s) - ‘packages.txt’ saved [1360/1360]

kernel:4.9.253-tegra
--2023-02-04 13:07:57--  https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/libcamera_dev_links.txt
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/26dffcaf-16dc-499f-9891-18a8b34bcb79?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230204%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230204T060757Z&X-Amz-Expires=300&X-Amz-Signature=aaccdbb850d03856061fedf94abbddf1c0259fe3401ed531fa349bb41ba7660f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera_dev_links.txt&response-content-type=application%2Foctet-stream [following]
--2023-02-04 13:07:57--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/26dffcaf-16dc-499f-9891-18a8b34bcb79?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230204%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230204T060757Z&X-Amz-Expires=300&X-Amz-Signature=aaccdbb850d03856061fedf94abbddf1c0259fe3401ed531fa349bb41ba7660f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera_dev_links.txt&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.109.133, 185.199.110.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4095 (4,0K) [application/octet-stream]
Saving to: ‘libcamera_dev_links.txt’

libcamera_dev_links 100%[===================>]   4,00K  --.-KB/s    in 0,001s  

2023-02-04 13:07:58 (4,20 MB/s) - ‘libcamera_dev_links.txt’ saved [4095/4095]

Cannot find the corresponding package, please send the following information to support@arducam.com
Hardware Revision: 
Kernel Version: 4.9.253-tegra
Package: libcamera_dev -- Bionic Beaver-arm64
```

Dion666, 2023-02-06:
You must be kidding!!!
The libcamera is for Raspberry Pi and not for Jetson device.

jasonrichdarmawan, 2023-02-06:
> You must be kidding!!! The libcamera is for Raspberry Pi and not for Jetson device.

No I am not. I am struggling with the gstreamer pipeline. I can't resize the nvarguscamerasrc 3840x2160 to 1280x1280 without having black pixels on the bottom of the camera.