ArduCAM/Arducam-Pivariety-V4L2-Driver #43: 0.0.12 breaks manual focus in libcamera

antisaint81, 2023-02-14:
With version 0.0.12 manual focus settings in libcamera no longer work correctly on the 64MP; the image is blurred no matter what it's set to. Blurry at zero and increasingly worse the higher you go. Using the manual focus python script works correctly, but libcamera by itself does not. I had to downgrade to 0.0.11 to fix it.

yaazzz, 2023-02-19:
Any update on that one?

SocMirror, 2023-02-20:
Same problem here, I had to retrograde version to be able to use my 64mp cam

loudjoudi, 2023-02-20:
> Same problem here, I had to retrograde version to be able to use my 64mp cam

Hi, Can you let me know which issue you have. I have no camera detected.

quitmeyer, 2023-03-22:
came here to say this. It seems like a simple fix, you just need to change the range of values from 0-15 instead of 0-1000 like you used to have it. cmon arducam folks :(

SocMirror, 2023-04-06:
@quitmeyer I will try that later on thank you.
but if the distance as decreased to 0-15, can we inject a float in order to be a bit more precise on the focus ?
Thank you 

If anyone else has managed to have a fix potentially.

quitmeyer, 2023-04-07:
yeah totally make it a float!