ArduCAM/Arducam-Pivariety-V4L2-Driver #47: IMX519 is confusing

surak, 2023-12-10:
The install_pivariety_pkgs.sh fails at the beginning, and all it does is to say that the driver is already there and does not need to be installed.

The git repo for the pivariety in [GitHub - ArduCAM/Arducam-Pivariety-V4L2-Driver: This driver is used for Arducam mipi camera with Pivariety board](https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver) points to [GitHub - ArduCAM/IMX519_AK7375](https://github.com/ArduCAM/IMX519_AK7375), but this fails completely to compile.

Could one show how to make this camera work on a clean install of Bookworm on a Pi 5?

Thanks!

Kocicak, 2024-03-08:
Hi, a bit late, but:

1. Go to https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/ and download latest .deb packages of:
* libcamera
* libcamera-dev
* libcamera-ipa
* rpicam-apps (in the next release libcamera-apps)
2. install these .deb packages manually via `sudo apt install ./libcam....deb

3. download file: https://raw.githubusercontent.com/ArduCAM/libcamera/arducam/src/ipa/rpi/pisp/data/imx519.json and replace this file: /usr/share/libcamera/ipa/rpi/pisp/imx519.json with the downloaded one

Optional step for faster and more reliable autofocus:
modify imx519.json file in /usr/share, in the end of file, there is section "rpi.af". Set these values:
"step_coarse": 1.0,
"step_fine": 0.025,
"step_frames": 1

Note that autofocus is not as good as with RPI 4.

surak, 2024-04-26:
Hi @Kocicak! Thanks for your reply. But I have some problems:

The drivers are for bullseye, not bookworm, which I am using. And the versions on raspberry os are way newer:

```
ii  libcamera-apps                        1.4.4-1                              all          transitional package
ii  libcamera-dev:arm64                   0.2.0+rpt20240418-1                  arm64        complex camera support library (development files)
ii  libcamera-ipa:arm64                   0.2.0+rpt20240418-1                  arm64        complex camera support library (IPA modules)
ii  libcamera-tools                       0.2.0+rpt20240418-1                  arm64        complex camera support library (tools)
ii  libcamera0.1:arm64                    0.1.0+rpt20231205-1                  arm64        complex camera support library
ii  libcamera0.2:arm64                    0.2.0+rpt20240418-1                  arm64        complex camera support library
ii  pipewire-libcamera:arm64              0.3.65-3+rpt7+deb12u1                arm64        PipeWire libcamera plugin
ii  python3-libcamera:arm64               0.2.0+rpt20240418-1                  arm64        complex camera support library (Python bindings)
```

When using lib camera on python, I get an error with the autofocus: 

```
[0:01:03.309755980] [1146]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:01:03.321946850] [1147]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[0:01:03.332979776] [1147]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx519@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[0:01:03.335104813] [1146]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[0:01:03.346805980] [1150]  INFO RPI pisp.cpp:695 libpisp version v1.0.5 999da5acb4f4 17-04-2024 (14:29:29)
[0:01:03.358139795] [1150]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx519@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[0:01:03.361374036] [1146]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
[0:01:03.361655054] [1146]  WARN V4L2 v4l2_pixelformat.cpp:344 Unsupported V4L2 pixel format RPBP
[0:01:03.362697610] [1146]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-XBGR8888 (1) 320x240-YUV420 (2) 1280x720-RGGB_PISP_COMP1
[0:01:03.362841461] [1150]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx519@1a - Selected sensor format: 1280x720-SRGGB10_1X10 - Selected CFE format: 1280x720-PC1R
[0:01:03.402068054] [1155]  WARN IPARPI ipa_base.cpp:1160 Could not set AF_TRIGGER - no AF algorithm or not Auto
```

Do you think this camera will ever get support on raspberry pi 5 / bookworm or we should stop buying arducam and go for the official ones?

Kocicak, 2024-04-27:
Hi, it is working in latest rpi os bookworm, I am using it myself. You have to download latest files from releases. Ignore packages from bullseye. Packages to install on bookworm are:

https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/libcamera-v0.0.5/libcamera-dev_0.2.0+rpt20240424-1_arm64.deb

https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/libcamera-v0.0.5/libcamera-ipa_0.2.0+rpt20240424-1_arm64.deb

https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/libcamera-v0.0.5/libcamera0.2_0.2.0+rpt20240424-1_arm64.deb

https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/libcamera-apps-v0.0.5/rpicam-apps_1.4.4-2_arm64.deb

these are the latest now. Download and install these 4 debs and autofocus will be working.

SamDecrock, 2025-05-06:
I'm trying to make it work on Bookworm today, but it does not find my camera.

It does work on Bullseye tough, but then I cannot compile it to work with MediaMTX: https://github.com/bluenviron/mediamtx-rpicamera/issues/48